warthog618/go-gpiocdev #1: Import Problem `github.com/warthog618/gpiod/device/rpi`

peteut, 2019-12-21:
I'm facing an issue using the `rpi` module, `go mod why` yields:

```
module github.com/warthog618/gpiod@latest found (v0.2.1), but does not contain package github.com/warthog618/gpiod/device/rpi
```

Import fragment is:

```golang
import (
        "fmt"

        "github.com/warthog618/gpiod/device/rpi"
)
```

Go version is `go version go1.13.4 linux/amd64`.

warthog618, 2019-12-21:
That would because that subpackage is not part of a tagged release yet as there was other code that I wanted to complete prior to tagging a release.
But, as I don't have time to complete that at the moment, I've tagged v0.3.0 so you can make use of what is there, including gpiod/device/rpi.

peteut, 2019-12-21:
@warthog618 just updated go to `go1.13.5` and `go tidy` resolved just fine. Thanks alot.

closed 2019-12-21 by peteut