warthog618/go-gpiocdev #2: Watcher Example Doesn't Compile?

samkalnins, 2020-02-09:
`RequestLine` doesn't seem to like the `gpiod.WithPullUp` option passed in (works fine without it):

```
l, err := c.RequestLine(offset,
		gpiod.WithPullUp,
		gpiod.WithBothEdges(func(evt gpiod.LineEvent) {
    ...
```

I get a 'invalid argument' panic.

warthog618, 2020-02-09:
Strangely it doesn't fail for me.
To be clear, you get an invalid argument panic when compiling?
Perhaps you are trying to run it on a Linux kernel < v5.5 where bias options are not supported?
If I run the example with kernel < 5.5 I get:

panic: invalid argument

goroutine 1 [running]:
main.main()
	/home/kent/work/go/gpiod/example/watcher/watcher.go:35 +0x32b

Is that what you are seeing?

If so, update your kernel if you want to use bias or remove option if you don't.
If not, please provide more information about what you are doing and your environment so I can try to reproduce it.

samkalnins, 2020-02-09:
Ah yes -- sorry, my kernel is <5.5 -- sorry, missed that requirement. Great library!

closed 2020-02-09 by samkalnins