warthog618/go-gpiocdev #3: collection of lines with different options

pseidemann, 2020-04-25:
hey,
first of all, nice library!

I wonder, is it possible to `RequestLines()`, while setting the option `gpiod.AsActiveLow` for a subset of these lines? I want all lines in one collection, so I can efficiently toggle values with `SetValues()`, but for some of these lines, I want `0` and `1` to be reversed.

warthog618, 2020-04-26:
The gpiod API follows the kernel UAPI which does not allow that - unfortunately all the requested lines in a set must be configured the same.
The simplest solution is to write a wrapper for `SetValues` that flips the polarity of select lines.

pseidemann, 2020-04-26:
too bad. thanks!

closed 2020-04-26 by pseidemann

warthog618, 2020-12-12:
FYI - collections of lines with different options is now supported in the GPIO uAPI v2 in Linux v5.10., which is supported by gpiod v0.6.0 using the *WithLines* config option.  Refer to the **Complex Configurations** section of the README for an example of how to setup a configuration as you requested.