warthog618/go-gpiocdev #4: feature request: support software serial

maltegrosse, 2020-04-27:
Hey, great work with this library. 
An amazing feature would be to have kind of software serial/software uart functionality, comparable to the arduino software serial library:
https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/SoftwareSerial/src/SoftwareSerial.cpp

This would allow us to use nearly any gpio for serial read and write.

what do you think?

warthog618, 2020-04-27:
That isn't a feature - that is a whole different layer, and so probably something best added to a separate library.

maltegrosse, 2020-04-27:
I thought it would be great having that "feature" on top of gpiod.
strictly speaking the spi drivers (and perhaps the devices) should also be in separate libraries, as they could also been seen as different layers.
My c++ skills are limited, so not sure if i could accomplish this in go...

warthog618, 2020-04-27:
"on top" means another library to me.

I am well aware that the SPI driver breaches that principle, and I was of two minds as to whether to include it within gpiod at all, but it is there as an example of what can be achieved via GPIO, and to demonstrate interfacing to hardware more complex than buttons and LEDs, not as a beachhead for additional drivers.

The driver you are suggesting is certainly achievable in Go.  In my experience C++ skills can get in the way of writing good Go, so you might be better placed than you think.  I would use that Arduino example only as guide - I certainly wouldn't try to port it directly to Go.  My greatest concern would be the accuracy of timings, given the driver would be running in Go on Linux - not exactly ideal for real-time.  That would limit the baud rates that you would be able to support.  That isn't as much of an issue for SPI with its separate clock line.

I would also seriously consider adding an actual UART, rather than entering into the world of pain involved in emulating one in software, but that might just be me....

maltegrosse, 2020-04-27:
Thank you for that hint.
I also came across https://github.com/adrianomarto/soft_uart 
which emulates the serial connection on system level - which is (for me) not really needed. but they also mention the issues about timings, real time etc - but also for that use case timing wouldnt be that problematic. 
The limitation of soft uart is that 1. its really limited to rpi, and you can not create multiple instances (what I need)
I think pigpio http://abyz.me.uk/rpi/pigpio/pigpiod.html also has some basic support. (read only)

maltegrosse, 2020-04-27:
okay, i think its a better and easier solution to solve this hardware wise - for example with 
https://rowlandtechnology.co.uk/epages/440a2cbf-3186-48ea-a54c-939a2c646da5.sf/en_GB/?ObjectPath=/Shops/440a2cbf-3186-48ea-a54c-939a2c646da5/Products/MULTIUART

thank you for the discussion and input.

closed 2020-04-27 by maltegrosse