warthog618/go-gpiocdev #5: getting "invalid argument" when calling RequestLine() with pull up

pseidemann, 2020-05-03:
hi,
when I execute the example https://github.com/warthog618/gpiod/blob/master/example/watcher/watcher.go on a raspberry pi, I'm getting "invalid argument" from `RequestLine()`. it seems this is caused by `gpiod.WithPullUp`, but when I remove it, it doesn't seem to work with a button which shorts to ground.

interestingly this call succeeds after I executed a python script which uses `RPi.GPIO` and sets the pull up:
```
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
GPIO.cleanup()
GPIO.setup(4, GPIO.IN, pull_up_down=GPIO.PUD_UP)
```

the problem returns after a reboot.

warthog618, 2020-05-03:
That sounds like you have an old kernel.  What version kernel are you running?  Note that the bias options require v5.5 or later.
RPi.GPIO uses a different approach (similar to my old gpio library) which is available in earlier kernels, but is deprecated and may be removed in later mainline kernels.

pseidemann, 2020-05-04:
ah yes that seems to be the reason. I use raspbian buster. I assume it will take quite some time until this api will work in (stable) raspbian? in the meantime I will probably migrate to your old api

warthog618, 2020-05-04:
Not sure when it will be in a release, but it is in the [raspbian kernel tree](https://github.com/raspberrypi/linux) if you are up to building your own kernel.

pseidemann, 2020-05-04:
maybe I will give it a shot. thanks!

closed 2020-05-04 by pseidemann