warthog618/go-gpiocdev #7: Import with GO modules not working

chetrf, 2020-07-26:
Let me start out by saying thanks for this library.  I am experimenting with it by just blinking an LED.  I am on an i.MX nano and it is in fact compiling and working correctly on the hardware.  But I have an editor issue that has me scratching my head.  I am using vscode on MacOs.  If edit the file as a local project I get errors on NewChip.  But if I edit the file directly on the i.MX nano using vscode remote SSH it figures things out and I get all the benefits of autocomplete etc.  The two images attached show what I am seeing.  So I doubt this is a actually an issue with the library, but maybe some pilot error on my part.  Any thoughts?  Thanks.
[image]
[image]

chetferry, 2020-07-27:
A bit more info.  I have been poking around trying to figure it out.  When I try to build it locally vs when I cross compile for linux you can see the results.  Not sure if this is helpful info?  Thanks.
[image]

warthog618, 2020-07-27:
That makes sense - gpiod only builds for linux.
Your editors are trying to compile it on OSX, and so are missing most of the files - the exception being options.go which is missing the +build linux and so tries to build.

You could setup your GOOS as per this VSCode issue - https://github.com/microsoft/vscode-go/issues/632

chetferry, 2020-07-27:
Good call on setting GOOS.  That cures it.  For any one else using vscode see the screen shot below.  Thanks for your help.
[image]

warthog618, 2020-07-28:
In light of this I will remove the build linux restrictions on the gpiod package files, but keep them in place for gpiod/uapi which is most definitely linux specific.  While you still wont be able to build binaries for OSX or Windows, gopls or whatever will be able to parse the user facing side of gpiod without needing to set GOOS.

closed 2020-10-07 by warthog618