warthog618/go-gpiocdev #8: cannot assign *gpiod.Line to l[i] (type gpiod.Line) in multiple assignment

alexlansovich, 2020-08-20:
Hello.
I need to assignment GPIO by type to map.
I have error - can you help?
i have maps in liftConfig.Sensors

BTN_MACHINE { 19 Кнопка Вызов в машинном помещении Нажата Не нажата 0 10 AsInput}
BTN_DOOR { 11 Дверь в лифтовую Закрыта Открыта 1 0 AsInput}
SW_SND_AMP { 24 Включение усилителя Выключено Включено 1 60 AsOutput}
SW_PUSK_ON { 18 Включение пускателя Выключено Включено 1 60 AsOutput}

`// start initial`
`	c, err := gpiod.NewChip("gpiochip0")`
`	if err != nil {`
`		panic(err)`
`	}`
`	defer c.Close()`
`	var l map[string]gpiod.Line`
`	// foreach sensors`
`	for i, value := range liftConfig.Sensors{`
`		fmt.Println(i,value)`
`		v := 0`
`		if value.GpioType == "AsOutput" {`
`			l[i], _ = c.RequestLine(value.IdSystem, gpiod.AsOutput(v))`
`		} else {`
`			l[i], _ = c.RequestLine(value.IdSystem, gpiod.AsInput)`
`		}`
`	}`

then i will foreach them and get state

`r, _ := l[i].Value()  // Read state from line (active / inactive)`

warthog618, 2020-08-20:
Try ` var l map[string]*gpiod.Line`

alexlansovich, 2020-08-20:
Thank you! It works!
You can write this in wiki

closed 2020-08-20 by alexlansovich