warthog618/go-gpiocdev #9: ACT LED support on Pi Zero

nullvariable, 2020-10-07:
Hi, reading the source it looks like GPIO 47 would throw an out of range error. On the Pi Zero, this is the raw pin for manipulating the onboard LED (marked ACT on the PCB).

Is it possible to access this pin via this library? Happy to test and put in a PR if it makes sense, but don't want to go down a rabbit hole if it's not doable.

nullvariable, 2020-10-07:
source: https://www.raspberrypi.org/forums/viewtopic.php?t=194262
Had this working with pigpio, but would prefer to use a Go native library like this.

nullvariable, 2020-10-07:
I got a bit further with this, I can confirm the pin is there, but I get `panic: device or resource busy` when I try to RequestLine
```
info, err := c.LineInfo(47)
fmt.Printf("%#v", info)
```
```
{
  Offset:47, 
  Name:"", 
  Consumer:"led0", 
  Requested:true, 
  IsOut:true, 
  ActiveLow:false, 
  OpenDrain:false, 
  OpenSource:false, 
  BiasDisable:false, 
  PullDown:false, 
  PullUp:false
}
```

warthog618, 2020-10-07:
The 
```
  Consumer:"led0", 
  Requested:true, 
```
part of the info means that pin is already requested by a kernel driver calling itself "led0", so it can't be requested from userspace using the GPIO uAPI - which is why you are getting the `device or resource busy` error.  

You can control that driver by writing to /sys/class/leds/led0/brightness as per [here](https://www.jeffgeerling.com/blogs/jeff-geerling/controlling-pwr-act-leds-raspberry-pi).

Other solutions, such as **pigpio** and my old **gpio** library, bypass the standard kernel uAPI and write directly to the Pi hardware, potentially conflicting with the kernel hardware drivers - something probably best avoided.

nullvariable, 2020-10-07:
ok cool, that makes a lot of sense. I don't generally need that LED, it was just a handy way to test code.

Thanks!

closed 2020-10-07 by nullvariable