warthog618/go-gpiocdev #10: panic: open /dev/gpiochip0: permission denied

soerenbrockmann, 2020-12-17:
I guess my issue doesn't belong directly to your great lib, but how can I get access to gpiochip0 ?

soerenbrockmann, 2020-12-17:
Found it:

sudo chmod a+rw /dev/gpiochip0 

or creating a build file and executing it with sudo

closed 2020-12-17 by soerenbrockmann

warthog618, 2023-05-25:
Alternatively, you can add your user to the group that owns the gpiochip, if there is one
e.g. on the Raspberry Pi it is the gpio group, so 
```
adduser you gpio
```
That can restrict the users allowed to access the gpio to a subset.

If the gpiochip is owned by root:root you have to go an extra step and create the group and chown the gpiochip.
And to get that to stick over reboot you probably have to add a udev rule to set the gpiochip ownership and permissions.
e.g. my RPi has
```
SUBSYSTEM=="gpio", GROUP="gpio", MODE="0660"
```
in /etc/udev/rules.d/99-com.rules.
It also has other GPIO rules, but they are only required for the deprecated sysfs interface.