warthog618/go-gpiocdev #11: DC Motor Control / PWM support

soerenbrockmann, 2020-12-17:
Hi, is there any example available of using your library for controlling a DC Motor using GPIO Pins:

The following code is an extract of a Python GPIO sysf implementation. I wonder if I can do the same with this gpiod library?
Especially controlling speed like with PWM....
    
   ```
 def forward(self, speed=1.0, duration=None):
        GPIO.output(self.left_motor[0],GPIO.HIGH)
        GPIO.output(self.right_motor[0],GPIO.HIGH) 
        GPIO.output(self.left_motor[1],GPIO.LOW)
        GPIO.output(self.right_motor[1],GPIO.LOW) 
        self.speed = ((speed - (-1))/2)*100
        self.pwm[0].ChangeDutyCycle(self.speed)
        self.pwm[1].ChangeDutyCycle(self.speed)
```

warthog618, 2020-12-17:
Short answer - No.

Not sure which Python library you are using, but they probably bundled the PWM and GPIO together in the one lib as they use the sysfs interface for both? The old GPIO sysfs interface is deprecated, btw.

This library deals only with GPIO character device interface that replaces the deprecated sysfs GPIO interface.  It does not provide any control over the PWM, which is a totally different interface- the sysfs interface for that is still current AFAIK.

There are Go packages that wrap the PWM sysfs interface, though I don't have experience with any of them so don't have any basis to make any recommendation.

soerenbrockmann, 2020-12-18:
They bundled it together, see here: https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio.py.

Yes, this lib makes use of the deprecated sysfs interface. Alright, I will do some more research regarding the PWM handling.

closed 2020-12-18 by soerenbrockmann