warthog618/go-gpiocdev #13: Jam hat single button press, but multiple handle calls.

krasi-georgiev, 2021-01-26:
I am trying to use this rpi hat on Rpi4
https://github.com/modmypi/Jam-HAT

```
g, err := gpiod.NewChip("gpiochip0")

g.RequestLine(rpi.GPIO19,
		gpiod.WithFallingEdge,
		gpiod.WithEventHandler(handleStart)
)
```

Problem is that most of the time it calls `handleStart` multiple times for a single press.
the problem is even worse if I press the button few times quickly - it will call the handle function few times more than the times I pressed the button.

I applied some throttling, but wondering if this is standard behaviour or you think it might be a bug?

warthog618, 2021-01-27:
You probably have an electrically noisy switch and need to debounce it (google "switch debounce").
(you would need to put a scope on the line to be sure, but it seems likely)
The library supports debouncing - with Linux v5.10 or later.  Use the WithDebounce option when requesting the line.
That kernel version is available for the Pi, but probably hasn't gotten to your distribution so you would need to built it yourself.

Alternatively you can debounce the button electrically, though that probably isn't an option if you are using a hat.  And you certainly seem to have a hat.

Btw, slower GPIO interfaces, such as the deprecated sysfs interface, tend to self-debounce as they drop events when they aren't handled fast enough - which is also a problem if that isn't what you want. The character device interface queues them so it is a lot less likely to lose critical events.  So another option is to use one of the slower APIs - they probably designed and tested the HAT using sysfs or Python.

Or you can perform the debouncing yourself in your handleStart.

krasi-georgiev, 2021-01-27:
Wow, amazing! thanks a lot, with an upgrade the kernel is exactly 5.10 so now with `gpiod.WithDebounce(100*time.Millisecond)` the problem is solved, Thanks Kent!

closed 2021-01-27 by krasi-georgiev