warthog618/go-gpiocdev #14: hang on line.Close() with edge handler

figadore, 2021-01-31:
I have 2 input lines and edge detection (`WithBothEdges`, `WithEventHandler`). When I attempt to close one line to prevent additional edge handling, the program hangs, nothing beyond `line.Close()` executes. Does the event handler need to be detached first or something?

warthog618, 2021-01-31:
Closing the line should be fine - nothing else required.
In what context are you calling the line.Close()?

warthog618, 2021-01-31:
Hmmm, if you are calling the line.Close() from within the handler try putting the close in a separate goroutine.
Otherwise the Close will deadlock waiting for the handler to complete.

figadore, 2021-01-31:
I wasn't seeing a deadlock error when I ran it with everything else, but when I tried to produce a minimal example, I started seeing the problem you suggest (`fatal error: all goroutines are asleep - deadlock!`).

```
package main

import (
	"fmt"
	"log"
	"os"
	"syscall"
	"time"

	"github.com/warthog618/gpiod"
)

const line = 8

func main() {
	chip, err := gpiod.NewChip("gpiochip0")
	if err != nil {
		panic(err)
	}
	replyCh := make(chan string)

	eventHandler := eventClosure(replyCh)
	btn, err := chip.RequestLine(line,
		gpiod.WithBothEdges,
		gpiod.WithEventHandler(eventHandler))
	if err != nil {
		fmt.Printf("RequestLine returned error: %s\n", err)
		if err == syscall.Errno(22) {
			fmt.Println("Note that the WithPullUp option requires kernel V5.5 or later - check your kernel version.")
		}
		os.Exit(1)
	}
	fmt.Printf("Watching Pin %d...\n", line)
	<-replyCh
	log.Printf("closing line")
	btn.Close()
	log.Printf("this statement is not reached")
}

func eventClosure(replyCh chan string) func(gpiod.LineEvent) {
	return func(evt gpiod.LineEvent) {
		t := time.Now()
		pressed := false
		if evt.Type == gpiod.LineEventFallingEdge {
			pressed = true
		}
		fmt.Printf("event:%3d %-7s %s (%s)\n",
			evt.Offset,
			pressed,
			t.Format(time.RFC3339Nano),
			evt.Timestamp)
		if evt.Offset == line && pressed {
			log.Printf("received 'end' button press, ending")
			replyCh <- "ok"
		}
	}
}
```

So `Close()` is not called from within the handler, I'm not sure why it is deadlocking

but when I change it to
```
	go btn.Close()
```

that appears to work. Is that what you were suggesting?

warthog618, 2021-02-01:
Hmmm, are you seeing two "received 'end' button press, ending"?

I suspect it is basically the problem I suggested with the Close() waiting for the event handler, but it is indirect as your event handler blocks in the `replyCh <- "ok"`, waiting for the main thread to read the replyCh.  That is assuming you are getting multiple interrupts and the second 'pressed' call to the event handler will block indefinitely, and so the Close() will never return.

Another way for the event handler to signal that the end has been pressed would be to close the channel rather than write to it - though it would have to check it hasn't already closed it previously, or a subsequent close will panic.

Alternatively, you could provide some way to force your event handler to exit for a controlled shutdown.  Typically I will provide a done channel that the main thread closes.  The event handler would select on either the `replyCh <- "ok"` or the done channel closing and exit when the done is closed.

I'm also suspecting the multiple interrupts are caused by bounce on your switch, so you could add debounce if you have a recent kernel (v5.10 or later).  Though you should still fix the potential deadlock anyway.

And you will only see the deadlock error if ALL of your goroutines are blocked, so when you run it with everything else there are still other goroutines active, so the runtime can't be sure you have a deadlock.

figadore, 2021-02-01:
Very clear explanation, thank you. Now that I understand why `Close()` is blocking waiting for the handler to complete, I'll work on finding a way to get the order correct. 

As for the debounce, unfortunately I'm on kernel 5.4, and I've had a bit of trouble because sometimes the rising edge event comes last, even after the last falling. Below are logs from a single button press where `false` is the last event, even though the button is pressed at this point (which makes the "unpressed" state override the "pressed" in the debounce function...)
```
event: 20 %!s(bool=false  ) 2021-01-31T16:02:25.111943101-08:00 (447816h2m25.111595105s)
event: 20 %!s(bool=false  ) 2021-01-31T16:02:25.121195001-08:00 (447816h2m25.114402074s)
...
event: 20 %!s(bool=false  ) 2021-01-31T16:02:25.17488542-08:00 (447816h2m25.118125034s)
event: 20 %!s(bool=true   ) 2021-01-31T16:02:25.179657368-08:00 (447816h2m25.118144034s)
event: 20 %!s(bool=true   ) 2021-01-31T16:02:25.183536326-08:00 (447816h2m25.118166034s)
event: 20 %!s(bool=true   ) 2021-01-31T16:02:25.187404284-08:00 (447816h2m25.118286032s)
event: 20 %!s(bool=false  ) 2021-01-31T16:02:25.192103234-08:00 (447816h2m25.121015003s)
```
not sure why, and it is difficult for me to even diagnose where the problem is without an oscilloscope or other fancy tools ( but maybe I can work around it by tracking state and only debouncing the falling edge when the last debounced rising or vice versa). If you think the problem might be in this repo, I can create an issue for it.

closed 2021-02-01 by figadore

warthog618, 2021-02-01:
I doubt the problem is in **gpiod** - it is literally just reading the events from the kernel.  The burst of events may be overflowing the event buffer in the kernel (which is only 16 events deep), though again you would need a recent kernel to confirm (the more recent kernel adds sequence numbers to the events).
The old kernel doesn't handle the overflow well either - it discards the most recent event which is almost always not what you want.  The more recent kernels discard the oldest event in the buffer, so once everything settles the final event is guaranteed to correspond to the current state.
So I would recommend upgrading to v5.10 and using the WithDebounce option, if that is at all possible.

Depending on your application you might also consider debouncing the switch electrically.

I will update the **gpiod** documentation to point out that the *Close* waits for the event handler to return, and that in general blocking in the event handler itself is a bad thing.