warthog618/go-gpiocdev #15: SetValues() not setting values as expected

tmelin, 2021-02-26:
Seeing some strange behaviour when trying to use gpiod on kernel 4.19.
SetValues() does not set lines at all, unless I also request another line as in example below.

`	chips := gpiod.Chips()
	log.Println(chips)

	c, err := gpiod.NewChip("gpiochip0")
	if err != nil {
		log.Println(err)
	}
	defer c.Close()

	l, err := c.RequestLines([]int{19, 20, 21}, gpiod.AsOutput(1, 1, 1))
	if err != nil {
		log.Println("RequestLines:", err)
	}
	defer l.Close()

	err1 := l.SetValues([]int{0, 1, 0})
	if err1 != nil {
		log.Println("SetValues:", err1)
	}
        // this must be here for SetValues() to have effect
	c.RequestLine(16, gpiod.AsOutput(1))

	rr := []int{0, 0, 0}
	l.Values(rr)
	log.Println("Values", rr)
`
With c.RequestLine(16...) in place, this yields expected output
2021/02/19 07:52:02 [gpiochip0 gpiochip1]
2021/02/19 07:52:02 Values [0 1 0]

But, if c.RequestLine(16...)  is removed, output is:
2021/02/19 07:50:38 [gpiochip0 gpiochip1]
2021/02/19 07:50:38 Values [1 1 1]

Can could be going on here? Am I doing something obviously wrong?

tmelin, 2021-02-26:
Adding lineInfo information from requested lines if that gives any additional clues
`2021/02/19 08:17:24 &{19 am0 gpiod-2045 true {false 2 0 0 0 false 0s 0}}
2021/02/19 08:17:24 &{20 am1_en gpiod-2045 true {false 2 0 0 0 false 0s 0}}
2021/02/19 08:17:24 &{21 am2_en gpiod-2045 true {false 2 0 0 0 false 0s 0}}`

Those lines are not used by any other application and are not gpio-hogged either.

warthog618, 2021-02-26:
That should work, so need to perform more debugging to try to identify the source of the problem, particularly whether the problem is in the kernel or gpiod.

Are you sure that it is not the get that is the problem - do the line voltages reflect the get or the set?

The c.RequestLine(16...) may just be providing a delay.  Have you tried anything else, e.g. a sleep?

What is your hardware platform?

Any chance you could try a more recent kernel?

tmelin, 2021-02-26:
looking at /sys/kernel/debug/gpio, output line states reflect the AsOutput() settings in RequestLines(). And they change as expected when setting 1/0 initial values. From debugfs, it's also visible that the lines are in fact in use by gpiod.
Tried with delay, but it seems output does not change regardless. So it would occur to me that it's setValues() call that is the problem. Also setValue() behaves likewise.

This is on ARM 32-bit platform. I could potentially try at some point with a more recent kernel, but don't have that set up ATM.

warthog618, 2021-02-27:
I have the reverse problem - testing on a recent kernel on ARM 32 doesn't show any problem, so I'll have to build a 4.19 kernel and see if I can replicate the behaviour you are seeing.

Looking at /sys/kernel/debug/gpio doesn't actually confirm the output value - it is still a get just at kernel level.  Does the physical value at the pin agree with the set or the get?

warthog618, 2021-02-28:
I've tried to repeat your issue on a Raspberry Pi Zero running Linux 4.19.97, and it doesn't appear for me.
I modified your example to get the line values between the first request and the set, and the second request, and then after the second request.

i.e.
```go
	l, err := c.RequestLines([]int{19, 20, 21}, gpiod.AsOutput(1, 1, 1))
	if err != nil {
		log.Println("RequestLines:", err)
	}
	defer l.Close()

	rr := []int{0, 0, 0}
	l.Values(rr)
	log.Println("Init Values", rr)

	err1 := l.SetValues([]int{0, 1, 0})
	if err1 != nil {
		log.Println("SetValues:", err1)
	}
	l.Values(rr)
	log.Println("Set Values", rr)

	// this must be here for SetValues() to have effect?
	c.RequestLine(16, gpiod.AsOutput(1))

	l.Values(rr)
	log.Println("Values", rr)
```

I get:
```
2021/02/28 20:39:39 Init Values [1 1 1]
2021/02/28 20:39:39 Set Values [0 1 0]
2021/02/28 20:39:39 Values [0 1 0]
```

which is exactly as expected - the set values are reported correctly even before the second request.

So I've gone as far as I can without knowing more about the particulars of your setup - there is obviously some critical difference between your setup and mine.
There could be something odd with the pinctrl (the kernel driver for particular GPIO hardware) for whatever hardware you are using that causes it not to report the correct values until a subsequent request?

tmelin, 2021-03-01:
Thanks for your effort!
Tested further that Requestlines() works correctly in all cases, physical line reflecting the actual request. It's setValues() that seems to not be going through.

Noticing that when requesting 19,20,21 lines, line 16 needs to be requested for it to work. If lines 10,11,12 instead are requested, then line 8 needs to be requested.

I'm also starting to suspect this is more related to the underlying driver rather than in user space side. The physical device is a I2C connected device, so there could be something in that communication that goes wrong.

tmelin, 2021-03-02:
Updated kernel to 5.10.17. Issue is no longer reproducible. I will now close this issue. Described problem related to kernel rather than gpiod.

Thanks for input and for doing this library!

closed 2021-03-02 by tmelin