warthog618/go-gpiocdev #18: Channels versus callback for event handlers?

entrope, 2021-03-24:
Should the functions `WithEventHandler()` and `WatchLineInfo()` use, or have equivalents that accept, application-created channels rather than callback functions?

The benefits:

- Channels are more idiomatic in Go.
- Channels make it easier to integrate into an application's event handling loop.
- Channels allow improved concurrency, and reduce the likelihood of accidentally blocking the event poller.

The drawbacks:

- A channel increases the latency of responding to an event.  (This is mitigated for some use cases by the timestamps in `LineEvent`, but those are more complicated for kernels v5.7-v5.10 if the application wants to map the event to a wall-clock time.)
- It constrains user code: A callback can easily write to a channel, but it is more complicated to emulate callbacks being triggered by a channel.

My personal use case matches cmd/gpiodctl/mon.go and cmd/gpiomon.go in having a callback that simply posts to a channel, so I am not sure how many real-world use cases have a strong preference for callbacks.

warthog618, 2021-03-24:
I've already answered this in a previous pull request.
No, due to point 1 of your drawbacks.  I don't to see how the app passing in a channel better than passing in a handler.

If the user wants channels then they can add them in the handler.  If the user wants to run in a goroutine then they can do it in their handler.  The handler callback is the lowest common denominator that allows the user to build a solution that best suits their use case, while keeping the gpiod code simple as well.

And I would dispute that using a channel in an API is more idiomatic - very few public APIs use channels.

entrope, 2021-03-25:
Thanks.  I suspected that might be the case.  I did not see an issue about it, and I did not think to check closed pull requests.

closed 2021-03-25 by entrope