warthog618/go-gpiocdev #19: Multi requestline in thread

vBlackOut, 2021-07-01:
hello,

 it's possible to use two requestLine for multithreading ? 

```
func pwm2defer(wg *sync.WaitGroup, percent2 uint32) {

	if os.Getenv("pwmSTOP") == "True" {
			defer wg.Done()
			os.Exit(0)
	}

  offset := rpi.GPIO20

	if ((percent2 >= 1) && (percent2 < 10000))  {

		time.Sleep((time.Duration(percent2) * time.Microsecond))
    // set value 0
    l, _ := c.RequestLine(offset, gpiod.AsOutput(1))
		time.Sleep((time.Duration(10) * time.Microsecond))
    // set value 0
    l.SetValue(0)
    l.Close()
	}

	defer wg.Done()

}

func pwm1defer(wg *sync.WaitGroup, percent1 uint32) {
	if os.Getenv("pwmSTOP") == "True" {
			defer wg.Done()
			os.Exit(0)
	}

  offset := rpi.GPIO16

	if ((percent1 >= 1) && (percent1 < 10000))  {
		time.Sleep((time.Duration(percent1) * time.Microsecond))
    // set value 0
    l, _ := c.RequestLine(offset, gpiod.AsOutput(1))
		time.Sleep((time.Duration(10) * time.Microsecond))
    // set value 0
    l.SetValue(0)
    defer l.Close()
	}

	defer wg.Done()

}

func eventHandler(evt gpiod.LineEvent) {

	var wg sync.WaitGroup
	wg.Add(2)
	go pwm1defer(&wg, percent1)
	go pwm2defer(&wg, percent2)
	wg.Wait()

}
```

thank you

warthog618, 2021-07-01:
Sure, why not?

vBlackOut, 2021-07-01:
device busy 
response : 

```
pwm1 10
debug pwm1: 8200

pwm2 10
debug pwm2: 8200

RequestLine returned error: device or resource busy
exit status 1

```

vBlackOut, 2021-07-01:
i upload code here : https://gist.github.com/vBlackOut/dbcd987d4a9b71da3cebb0066e3606fc

warthog618, 2021-07-01:
In your pwm() you defer l.Close() inside the loop, so the second iteration will see that error.
Without that defer I get:
```
./vBlackOut 
init pwm process...
please wait command usage :
 cmd : pwm[1-2] percent[0-100%]
 cmd : [reset] for disable all comand 
 cmd : [stop] for quit program 
pwm1 10

debug pwm1: 8200
pwm2 10

debug pwm2: 8200
reset

pwm1 10

debug pwm1: 8200
```

You shouldn't request and release lines continually like that - it is better to request them once and hold them and just toggle them in the worker goroutines.  Have the main() close them as it exits.

Also, in pwm1defer(), your l.Close() and wg.Done() calls will be called in the reverse order because they are both deferred.  That means eventHandler() could exit and get called again before the l.Close() gets called - that could also cause the resource busy you are seeing.

Why defer calls when you are about to exit?  Put the defer wg.Done() once at the beginning of the goroutine.  Or put wg.Done() before each exit.  Not defer just before you exit.

vBlackOut, 2021-07-01:
Ho thank you it's interesting !! just error in beginning to golang sorry :) thank you so much !

vBlackOut, 2021-07-01:
i retested but the code if not defer the led not detected in eventhandler

vBlackOut, 2021-07-01:
eventhandler after the loop and close after the loop thank you !! Ho i found other the solution i implemented just the

closed 2021-07-01 by vBlackOut