warthog618/go-gpiocdev #20: Test Errorf format %w has unknown verb w

vBlackOut, 2021-07-04:
i launch this test but return error ? why 

```
GOOS=linux GOARCH=arm GOARM=7 go test -c
# github.com/warthog618/gpiod_test
./gpiod_test.go:1073: Errorf format %w has unknown verb w
```

warthog618, 2021-07-05:
Your test run is failing as either your kernel doesn't have the gpio-mockup module, or the user you are running as doesn't have rights to load the gpio-mockup module.
The error being reported is because that test is using a feature (%w) added in Go 1.13, and I can only assume you are using an earlier version of Go (1.12??).
I've updated go.mod to require Go 1.13 or later.

When I run the test, compiled with Go 1.16.5, on my 4B as a user that doesn't have module loading rights I get:
`Platform not supported - error loading gpio-mockup: failed to load gpio-mockup: exit status 1`

vBlackOut, 2021-07-05:
ok i understand i try this week thank you :)