warthog618/go-gpiocdev #23: can't install

reejinbouk, 2021-09-01:
Hi
tried to install, got this:

# github.com/warthog618/gpiod
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/gpiod.go:306:30: undefined: uapi.LineInfo
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/gpiod.go:331:32: undefined: uapi.LineInfoV2
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/gpiod.go:371:21: undefined: uapi.LineInfo
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/gpiod.go:381:23: undefined: uapi.LineInfoV2
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/gpiod.go:574:38: undefined: uapi.HandleFlag
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/gpiod.go:607:37: undefined: uapi.EventFlag
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/gpiod.go:620:44: undefined: uapi.LineFlagV2
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/gpiod.go:654:50: undefined: uapi.LineAttribute
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/options.go:93:45: undefined: uapi.OutputValues
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/options.go:101:29: undefined: uapi.LineConfigAttribute
../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/options.go:101:29: too many errors

warthog618, 2021-09-01:
I don't get that, so you will need to provide more detail as to what you are doing to create that result.

Does ../../pkg/mod/github.com/warthog618/gpiod@v0.6.0/uapi/ exist and correspond to the repo?

reejinbouk, 2021-09-03:
tried on linux - works
close

Arcnor, 2021-09-03:
I'm also getting the same error, both with v0.6.0 and the current tip. I'm trying to compile on macOS and the `uapi` folder exists (I'm not sure how to check if it "corresponds to the repo", but given it was just downloaded and it's full of files I'm assuming it is?).

Do you have any more ideas about what to check? My knowledge of Go is near 0 unfrotunately, but I'd like to use this library for some experiments, so I'm happy to test whatever you think can help.

Thanks in advance.

closed 2021-09-04 by warthog618

warthog618, 2021-09-04:
So the problem here is trying to perform `go get` on platforms other than Linux and the **uapi** subpackage only builds for Linux (building for other platforms makes no sense).

The simplest solution is to tell `go get` not to compile the package:
```
go get -d github.com/warthog618/gpiod
```
The package will be compiled as appropriate when you cross-compile for Linux.

Arcnor, 2021-09-04:
Thanks, that's actually a better solution, and will probably fix a potential issue I was thinking about (cross compiling for multiple platforms).