warthog618/go-gpiocdev #24: watcher issue

Bobby-88, 2021-09-10:
Hello and thank you for the great library.
I'm playing with the wiegand reader and watcher form the example directory. I use 5V to 3.3V level shifter to feed into the RPI 3B+.
with a slightly modified version of the watcher.go from the examples directory i'm seeing this output:
```pi@raspberrypi:~/gowsclient $ go run watcher.go
Watching Pin 17, 18...
event: 17 rising  2021-09-10T17:41:12.837029332+03:00 (49m33.58660895s)
event: 17 rising  2021-09-10T17:41:12.840595984+03:00 (49m33.590528309s)
event: 18 rising  2021-09-10T17:41:12.840554578+03:00 (49m33.590490132s)
event: 18 falling 2021-09-10T17:41:12.842083478+03:00 (49m33.590492632s)
event: 18 rising  2021-09-10T17:41:12.842545924+03:00 (49m33.592475853s)
event: 18 falling 2021-09-10T17:41:12.842595091+03:00 (49m33.592478301s)
event: 18 falling 2021-09-10T17:41:12.844557114+03:00 (49m33.594513554s)
event: 17 rising  2021-09-10T17:41:12.842551654+03:00 (49m33.59251403s)
event: 18 falling 2021-09-10T17:41:12.844624978+03:00 (49m33.594515116s)
event: 18 falling 2021-09-10T17:41:12.846503512+03:00 (49m33.596429275s)
event: 17 rising  2021-09-10T17:41:12.846527002+03:00 (49m33.596466098s)
event: 18 falling 2021-09-10T17:41:12.846547314+03:00 (49m33.596431723s)
event: 18 rising  2021-09-10T17:41:12.850367872+03:00 (49m33.600295874s)
event: 17 rising  2021-09-10T17:41:12.850379903+03:00 (49m33.600332905s)
event: 18 rising  2021-09-10T17:41:12.850413132+03:00 (49m33.60029827s)
event: 18 falling 2021-09-10T17:41:12.850458809+03:00 (49m33.600301238s)
event: 18 rising  2021-09-10T17:41:12.850524329+03:00 (49m33.600303738s)
event: 18 rising  2021-09-10T17:41:12.85064261+03:00 (49m33.600306134s)
event: 18 rising  2021-09-10T17:41:12.850687298+03:00 (49m33.600308738s)
event: 17 rising  2021-09-10T17:41:12.850584277+03:00 (49m33.600492175s)
event: 18 rising  2021-09-10T17:41:12.850732974+03:00 (49m33.600311238s)
event: 18 falling 2021-09-10T17:41:12.850820422+03:00 (49m33.600313634s)
event: 18 falling 2021-09-10T17:41:12.85090938+03:00 (49m33.600316082s)
event: 18 rising  2021-09-10T17:41:12.850951463+03:00 (49m33.600318582s)
event: 18 rising  2021-09-10T17:41:12.850992401+03:00 (49m33.600321082s)
event: 18 rising  2021-09-10T17:41:12.851033077+03:00 (49m33.60032353s)
event: 18 rising  2021-09-10T17:41:12.851073806+03:00 (49m33.600326551s)
event: 18 rising  2021-09-10T17:41:12.8511149+03:00 (49m33.600328947s)
event: 18 rising  2021-09-10T17:41:12.851155577+03:00 (49m33.600331394s)
event: 18 falling 2021-09-10T17:41:12.851196462+03:00 (49m33.60033379s)
event: 18 rising  2021-09-10T17:41:12.851237139+03:00 (49m33.600496029s)
event: 18 falling 2021-09-10T17:41:12.852337655+03:00 (49m33.602319876s)
event: 18 falling 2021-09-10T17:41:12.852424999+03:00 (49m33.602339303s)
event: 18 rising  2021-09-10T17:41:12.852499946+03:00 (49m33.602469927s)
event: 18 falling 2021-09-10T17:41:12.852572655+03:00 (49m33.602540969s)
event: 18 rising  2021-09-10T17:41:12.85264604+03:00 (49m33.602615031s)
event: 18 rising  2021-09-10T17:41:12.8527188+03:00 (49m33.602689145s)
event: 18 rising  2021-09-10T17:41:12.852792549+03:00 (49m33.602761072s)
event: 18 falling 2021-09-10T17:41:12.852865987+03:00 (49m33.602835082s)
event: 18 rising  2021-09-10T17:41:12.852940882+03:00 (49m33.602911176s)
event: 18 rising  2021-09-10T17:41:12.853014528+03:00 (49m33.602985082s)
event: 18 rising  2021-09-10T17:41:12.853088017+03:00 (49m33.603056956s)
event: 18 rising  2021-09-10T17:41:12.853160673+03:00 (49m33.60313081s)
event: 18 rising  2021-09-10T17:41:12.853234214+03:00 (49m33.603204612s)
event: 18 rising  2021-09-10T17:41:12.853307443+03:00 (49m33.603278362s)
event: 18 rising  2021-09-10T17:41:12.85338036+03:00 (49m33.603349611s)
event: 18 rising  2021-09-10T17:41:12.853452703+03:00 (49m33.603423674s)
event: 18 falling 2021-09-10T17:41:12.853536609+03:00 (49m33.603495184s)
event: 18 falling 2021-09-10T17:41:12.853609473+03:00 (49m33.603580027s)
event: 18 rising  2021-09-10T17:41:12.853737858+03:00 (49m33.603706068s)
event: 18 rising  2021-09-10T17:41:12.853811191+03:00 (49m33.603780078s)
event: 18 rising  2021-09-10T17:41:12.853884264+03:00 (49m33.603854089s)
event: 18 falling 2021-09-10T17:41:12.85395942+03:00 (49m33.603930338s)
event: 18 rising  2021-09-10T17:41:12.854032388+03:00 (49m33.604001536s)
event: 18 rising  2021-09-10T17:41:12.854104992+03:00 (49m33.604075338s)
event: 18 rising  2021-09-10T17:41:12.85417796+03:00 (49m33.604149087s)
event: 18 rising  2021-09-10T17:41:12.854250721+03:00 (49m33.604220441s)
event: 18 rising  2021-09-10T17:41:12.854323741+03:00 (49m33.60430367s)
event: 18 falling 2021-09-10T17:41:12.854396658+03:00 (49m33.604374659s)
event: 18 rising  2021-09-10T17:41:12.85446973+03:00 (49m33.60443919s)
event: 18 rising  2021-09-10T17:41:12.854542334+03:00 (49m33.604512888s)
event: 18 falling 2021-09-10T17:41:12.854616188+03:00 (49m33.604586794s)
event: 18 rising  2021-09-10T17:41:12.854688583+03:00 (49m33.604658252s)
event: 18 falling 2021-09-10T17:41:12.854761344+03:00 (49m33.604732002s)
event: 18 rising  2021-09-10T17:41:12.854833999+03:00 (49m33.604803512s)
event: 18 falling 2021-09-10T17:41:12.854906447+03:00 (49m33.604877418s)
event: 18 rising  2021-09-10T17:41:12.854978947+03:00 (49m33.604948719s)
event: 18 rising  2021-09-10T17:41:12.855053217+03:00 (49m33.605022469s)
event: 18 rising  2021-09-10T17:41:12.855126446+03:00 (49m33.605095542s)
event: 18 falling 2021-09-10T17:41:12.855199363+03:00 (49m33.605169656s)
event: 18 rising  2021-09-10T17:41:12.855271654+03:00 (49m33.605241114s)
event: 18 falling 2021-09-10T17:41:12.855344258+03:00 (49m33.605314864s)
event: 18 rising  2021-09-10T17:41:12.855416758+03:00 (49m33.605386218s)
event: 18 rising  2021-09-10T17:41:12.855489622+03:00 (49m33.605459915s)
event: 18 falling 2021-09-10T17:41:12.855562538+03:00 (49m33.605533561s)
event: 18 falling 2021-09-10T17:41:12.855634986+03:00 (49m33.605604915s)
event: 18 falling 2021-09-10T17:41:12.855802589+03:00 (49m33.605771216s)
event: 18 falling 2021-09-10T17:41:12.855876652+03:00 (49m33.60584606s)
event: 18 rising  2021-09-10T17:41:12.855948943+03:00 (49m33.605919861s)
event: 18 falling 2021-09-10T17:41:12.856021703+03:00 (49m33.605991111s)
event: 18 rising  2021-09-10T17:41:12.856095661+03:00 (49m33.606065017s)
event: 18 rising  2021-09-10T17:41:12.856168213+03:00 (49m33.606138506s)
event: 17 rising  2021-09-10T17:41:12.8562779+03:00 (49m33.606226423s)
event: 18 rising  2021-09-10T17:41:12.856240608+03:00 (49m33.606189964s)
event: 18 falling 2021-09-10T17:41:12.856410295+03:00 (49m33.606227308s)
event: 18 rising  2021-09-10T17:41:12.856482847+03:00 (49m33.606453661s)
event: 18 falling 2021-09-10T17:41:12.856555295+03:00 (49m33.606525224s)
event: 18 rising  2021-09-10T17:41:12.856627117+03:00 (49m33.606596629s)
event: 18 rising  2021-09-10T17:41:12.856698836+03:00 (49m33.606667723s)
event: 18 rising  2021-09-10T17:41:12.856770919+03:00 (49m33.606739285s)
event: 18 falling 2021-09-10T17:41:12.856843054+03:00 (49m33.606813035s)
event: 18 rising  2021-09-10T17:41:12.856914616+03:00 (49m33.606884441s)
event: 18 falling 2021-09-10T17:41:12.856987793+03:00 (49m33.606958347s)
event: 18 falling 2021-09-10T17:41:12.857060084+03:00 (49m33.607029701s)
event: 18 rising  2021-09-10T17:41:12.857134094+03:00 (49m33.607103607s)
event: 18 rising  2021-09-10T17:41:12.857206125+03:00 (49m33.607177356s)
event: 18 rising  2021-09-10T17:41:12.85727826+03:00 (49m33.607248814s)
event: 18 rising  2021-09-10T17:41:12.857350343+03:00 (49m33.607320116s)
event: 18 rising  2021-09-10T17:41:12.857422999+03:00 (49m33.607393866s)
event: 18 rising  2021-09-10T17:41:12.857495187+03:00 (49m33.607465324s)
```
the question is - how is it possible to more than 1 event consecutively? I thought once there is a transition to lower state (fallingEdge) - there should be only risingEdge after that. But i see here in the output like 6 consecutive rising events. 
This leads to the issue that i'm unable to correctly read the data from wiegand. I've tested with logic analyzer and there everyting shows up correctly:

[image]

please give me some hints

warthog618, 2021-09-11:
There are several ways you could lose events.  The line may change state faster than the Pi can handle, so while it gets the interrupt from the line going low by the time the kernel checks the line level it is back high.  And there is a FIFO of events in the kernel, that is only 16 events deep, which you could be overflowing - you have over 50 events in the plot.
If you are running a recent kernel supporting GPIO uAPI v2 then you will also have sequence numbers on the events so you can check if the FIFO is overflowing. You can also increase the size of the FIFO via the uAPI using the WithEventBufferSize(num) option.
And, while the scope lines look clean in this plot that depends on your scope so you may still be getting noise that needs to be debounced.  If the event list corresponds to the plot then you have more events than you should so a lot of those are spurious noise.  You can debounce the lines using with WithDebounce(period) - again that requires a recent kernel.

Bobby-88, 2021-10-01:
Thank you warthog.
Over last few days I've:
1. tried to check the seq numbers - i have kernel version 5.10.60-v7+, but i'm not really sure how those sequence numbers are exposed? Also i'm not really sure which version of kernel should support this new API - google shows me a lot of mailing lists with you involved :) Truly amazing stuff, all this low level linux drivers/kernel is a black magic for me. Thank you much for your effort.
2. Using WithEventBufferSize also didn't help - e.g. i was receiving exactly same output (well, randomized garbage, to be accurate).
3. WithDebounce also did not help - the results were the same

I got another logic analyzer and here is what i've got:
[image]
the issue was that the logic converter (i was using TI TXS0108E) was producing this noise on the output. 
I've replaced TXS0108E (well this is surely workaround, but it's tomorrows problem) with other logic converter, added 470pF capacitor to the ground on the signal wire and now i see the results with ~98% consistency:

```
binary: 000100001011111100001010
int: 1097482
card id: 10bf0a
```
so, to sum up, i had hardware issues which i failed to identify in the first place.

thank you for the support.

warthog618, 2021-10-02:
>     1. tried to check the seq numbers - i have kernel version 5.10.60-v7+, but i'm not really sure how those sequence numbers are exposed? Also i'm not really sure which version of kernel should support this new API.

The LineEvent passed to the event handler contains the sequence numbers. Ooops, the sequence numbers are only exposed in master not v0.6.0.  My bad, I've obviously got more minor changes built up in master than I thought - I should tag a new release in near future.  The examples don't use them yet as they haven't been updated for v2 either.
For old kernels that don't support the new uAPI they will always be 0.  That would be for all kernels before v5.10, so you should be using uAPI v2.  You can call UapiAbiVersion() on the chip or line to check which version the library is using.

> 
>     2. Using WithEventBufferSize also didn't help - e.g. i was receiving exactly same output (well, randomized garbage, to be accurate).

The WithEventBufferSize is only a hint for the kernel - the FIFO sizes in the kernel are sized in powers of 2, so you need to double the size, or more, to see any difference - if you were getting any overflow.

> 
>     3. WithDebounce also did not help - the results were the same
> 

That is odd, as the debounce should filter out events occurring more frequently than the period, well IIRC strictly the line has to stop oscillating for the period, and I've tested that on several RPi.  What periods were you using?

> so, to sum up, i had hardware issues which i failed to identify in the first place.
> 
> thank you for the support.

Glad you're making progress.

closed 2022-01-20 by warthog618