warthog618/go-gpiocdev #29: Software PWM

maltegrosse, 2022-04-17:
Hi,
Is it possible to include software pwm handling using non pwm pins as a pwm replacement?
E.g
https://github.com/WiringPi/WiringPi/blob/master/wiringPi/softPwm.c

warthog618, 2022-04-17:
Yes and no.
It is trivial enough to do it in Go using gpiod to drive the pins.
But PWM is conceptually a separate API and I would rather keep the scope of this library to strictly GPIO.

maltegrosse, 2022-04-17:
So e.g. adopting the wiring pi in go would be trivial? I don’t plan to use real pwm pins, just use normal gpios to control a fan, as my board does not have pwm headers but a lot of spare gpios

warthog618, 2022-04-17:
I repeat - It is trivial enough to do it in Go using gpiod to drive the pins.
You can easily write a goroutine that does what the wiringpi code does that in softPwmThread.

maltegrosse, 2022-04-17:
All right, than I will try it out once I got some time and will provide some example here

closed 2022-04-17 by maltegrosse

Klaus-Tockloth, 2024-01-23:
@maltegrosse : Did you find a solution for software generated PWM?

maltegrosse, 2024-01-23:
@Klaus-Tockloth I have not continue working on this - sorry.

warthog618, 2024-01-24:
@maltegrosse @Klaus-Tockloth There is work being done on a [kernel driver](https://lore.kernel.org/linux-gpio/20210117130434.663qpp6noujptdyt@pengutronix.de/) for this, that would take any GPIO and present it to userspace as a PWM.  It is still a work in progress, so it hasn't hit the kernel yet, but I would be looking at using that when it does.

Klaus-Tockloth, 2024-01-24:
Implementation in a kernel driver ... that sounds promising. Should we open a separate issue for this? Probably others are also interested in software generated PWM for any pin.

BTW: I have tried to write a Golang package for software generated PWM. The solution is based on go routines and time.Sleep(). But I ran into limitations (probably with the unsufficient resolution of time.Sleep()).

warthog618, 2024-01-24:
@Klaus-Tockloth The timer resolution, particularly using time.Sleep(), and context switching overhead would definitely limit what you can do from userspace.

I think this issue is sufficient, and the driver itself is out of the scope of this library.

I must chase up where that driver is at when I have a chance - seems quite a few people are interested in it, particularly with the Pi community moving towards standard Linux interfaces.