warthog618/go-gpiocdev #30: RequestLine returns "no such file or directory" error

benni-tec, 2022-09-05:
Hey,
so I am able to open the chip using `gpiod.NewChip("gpiochip0")`. This does not return an error and the returned `*gpiod.Chip` is non-nil, however when trying to open a line `l := c.RequestLine(0, ...)` the error `no such file or directory` is returned. 

I've looked into your code a little bit and while I don't claim to understand it, to me it seems like this shouldn't be able to happen since the subsequent calls from `chip.RequestLine()` all use the file descriptor of the already opened/present file used to open the chip.

Maybe I overlooked something, but I was not able to identify the cause of this issue. Any help would be greatly appreciated!

go version go1.19 linux/amd64
github.com/warthog618/gpiod v0.8.0
ubuntu server(minimized) 5.15.0-47-generic 

---

Below is the code from my project, however I don't think there in lies the problem, I will include it for good measure though!
```
g.chip, err = gpiod.NewChip("gpiochip0")
	if err != nil {
		return nil, err
	}

	println("opened chip", g.chip.Label)

	var es []error
	var e error

	request := func(offset int, op ...gpiod.LineReqOption) (l *gpiod.Line) {
		l, e = g.chip.RequestLine(offset, op...)
		if e != nil {
			es = append(es, e)
		}

		return l
	}

	g.Light = request(0, gpiod.AsOutput(1))
	g.Brightness = request(1, gpiod.AsOutput(1))
	g.Button = request(2, gpiod.WithDebounce(time.Millisecond*150), gpiod.WithEventHandler(buttonHandler), gpiod.WithPullUp)

	if len(es) != 0 {
		err = fmt.Errorf("%v", es)
	}
```

This returns the following:
```
opened chip mcp2221_gpio
gpio init error: [no such file or directory no such file or directory no such file or directory]
```

warthog618, 2022-09-06:
That error (ENOENT which the error formatter has printed as "no such file or directory") comes from the kernel and could indicate that those lines are not assigned to GPIO in the kernel.

What does gpioinfo (in the gpiod/cmd/gpioinfo directory) show for those lines?
Can you access the lines with the gpioset or gpioget in the corresponding cmd directory?
If those work then reduce your example to the simplest possible code that fails.  So only one line, no request closure, and no error list.  Just the gpiod calls.

benni-tec, 2022-09-06:
First of all thank you for the quick response! gpioinfo does recognizet he chip and all 4 lines, however gpioget/set throws the same error:
```
fotobox@fotobox:~$ gpioinfo
gpiochip0 - 4 lines:
        line   0:      unnamed       unused   input  active-high
        line   1:      unnamed       unused   input  active-high
        line   2:      unnamed       unused   input  active-high
        line   3:      unnamed       unused   input  active-high
fotobox@fotobox:~$ gpioget /dev/gpiochip0 0
gpioget: error reading GPIO values: No such file or directory
```

After a bit more googeling I found this patchwork.kernel.org [thread ](https://patchwork.kernel.org/project/linux-input/patch/20210818152743.163929-2-tobias.junghans@inhub.de/#24414601), apparently the problem lies with the driver not properly setting up the mcp2221(a) chip I'm using (on an adafruit breakout board).

Sorry for bothering you, apperently the patch didn't make it into the ubuntu LTS version yet. (Altough I was unable to track it 😓/have no experience finding patches in diffrent linux flavours )

benni-tec, 2022-09-06:
For future reference: While it is possible to manually configure the MCP2221 chip to use the GP pins as GPIO, I found it much easier to download the "MCP2221 Utility" from the [manufactures website](https://www.microchip.com/en-us/product/MCP2221A) and configuring it on a Windows PC.

Note: I was not able to use the `gpiod.WithDebounce()` Option as this would result in a ENOENT error as well. Since my kernel version is greater than v5.10 I'm guessing this is a limitation of the chip, and I will have to implement debounce manually.

Apart from this everything now works just fine, so I will be closing this issue! Thank you for your help

closed 2022-09-06 by benni-tec

warthog618, 2022-09-06:
> Note: I was not able to use the `gpiod.WithDebounce()` Option as this would result in a ENOENT error as well. Since my kernel version is greater than v5.10 I'm guessing this is a limitation of the chip, and I will have to implement debounce manually.

The kernel should perform the debouncing for you, whether the driver supports it or not (it falls back to software debouncing), so this sounds like another bug in the kernel.

benni-tec, 2022-09-06:
Yes you are right. According to the datasheet/manufacturers information the MCP2221(a) should work with the generic usb HID/CDC_ACM drivers and therefore gpiod sysfs etc, however I‘m **guessing** due to some improper implementation (I think I remember something about it also not properly supporting some CDC_ACM protocol) there are probably some bugs/incompatibility’s with those drivers. 

While the output now works fine a input configured with EdgeDetection  also throws the ENOENT error and I haven’t been able to manually detect edges/state changes. However given the other problems I think this is a kernel/driver/chip problem and not caused by your library.

warthog618, 2022-09-06:
Ah, the problem isn't debounce but edge detection.  If driver doesn't correctly support interrupts - that would explain it.  As I said - kernel bug, or maybe hardware limitation.
And I have a side interest in the source of the problem, as I wrote the GPIO uAPI that was released in Linux v5.10 and was a bit concerned that there might be a corner case for the debounce fall back that I'd missed. But a problem in the driver or hardware looks much more likely, and that is outside my scope.

warthog618, 2022-09-06:
Probably another driver limitation. Note that for historical reasons setting bias (pull up  or down) in the kernel is best effort, so if the driver does not support bias then the setting is effectively ignored - which is probably what is happening here.  I did look into changing that behaviour to propagate an error, but it turns out it would be a lot of work for relatively little gain, and so let it be.
I assume gpioget reads the line correctly if you pull it externally?