warthog618/go-gpiocdev #34: Concurrency with context and select

purpleidea, 2022-10-13:
Nice library, thanks!
I was reading through some of the code and in particular around:

```
	// from the watcher example
	offset := rpi.J8p16
	l, err := gpiod.RequestLine("gpiochip0", offset,
		gpiod.WithPullUp,
		gpiod.WithBothEdges,
		gpiod.WithEventHandler(eventHandler))
	if err != nil {
		...
	}
	defer l.Close()
```

Once the l.Close() returns, does this guarantee that future events (calls to eventHandler) *won't* occur?
Assuming that's plumbed correctly, I was thinking of wrapping this in a more standard context style API, eg: https://pkg.go.dev/context
I think the most standard way is a function that you call that blocks until there is an event or the context cancels. Event handlers/callbacks are a bit more difficult to use I think.

Are you interested in having such helper functions upstream if I build that?
That way the context can cancel things, and I can wrap the blocking function so it sends on a channel of events so I can use select{} on it directly.

eg:

```
ctx := context.TODO()
// setup()
for {
	result, err := BlockingFunc()
	if err != nil {
		return ...
	}
	select {
	case event <- struct{...}:
	case <-ctx.Done():
		return ...
	}
}
```

Thanks for this great lib, it was pretty much exactly what I was looking for.
I only wish the equivalent was possibly in tinygo and that ESP32 and similar had ethernet support! Going to go heavy with rpi in the meantime!

warthog618, 2022-10-14:
Checked in a select_watcher example to demonstrate how to setup a context pattern.
That is quite easy to do, and gives the user complete control over how they want to manage the event chan, so again not sure there is any benefit trying to integrate that into the gpiod itself.

Whether event handlers or channels are easier to use depends on your application, but there is no argument that the event handler provides the simpler interface, and that the user can extend that into a context pattern with little effort if that suits.

I'm sure I've had a variation of this argument before in another issue - ah issue #18 and pr #12.

purpleidea, 2022-10-17:
I don't recommend a channel in the actual API, but rather a blocking function that closes with the context it's called with cancelling...

In any case, I'm pretty sure I can build that out from your event handler API-- it's just transforming the API, not doing something fundamentally intractable. You already did the hard part by talking to the kernel!

I will post my code as soon as it's done. Got a few projects in the air, but your lib unblocks one of mine!

Thanks again

warthog618, 2022-10-17:
The blocking function is the select in the select_watcher example.  But if you have a simpler way to set that up then I'm happy to take a look at it.