warthog618/go-gpiocdev #35: gpiod_line_event_wait_bulk with IO on different gpiochips

Vernon-Naidoo-1, 2023-01-19:
I'm porting some code that uses gpiod_line_event_wait_bulk, however on the new base, the gpio are on different gpiochips. There is a check in gpiod_line_event_wait_bulk that will return -1 if !line_bulk_same_chip(bulk), so this always fails. However, I patched out that check and ran my code and it works. What's the reason behind this check? If it isn't safe to remove this check, is there an alternate way to do this for multiple IO on different gpiochips?

warthog618, 2023-01-20:
Are you writing in Go, cos your question seems to be focused on libgpiod C functions.  This library is a Go version of libgpiod.
You should direct libgpiod questions to linux-gpio@vger.kernel.org, as per  the libgpiod README.

Anyway, you say your code runs and works.  Unfortunately you are mistaken.  Your code will probably run (depending on the offset ranges), but it isn't doing what you want it to do.  The API used by libgpiod strictly operates on individual chips, and the libgpiod API does not support requesting bulk lines across multiple chips.  It will instead use the offsets for the bulk lines on one of the chips.  This is certainly not what you want.  If you were using the bulk for sets it may even damage your hardware depending on what those lines connect to.  So it is NOT safe to remove that check.

If you have lines on separate chips then you need a separate request for each chip, and use select or epoll or whatever to merge the events from those requests into a single stream.  If you need an example, the gpiomon for libgpiod v2 does this (not released yet, but in libgpiod master).  That requires the lines be identified by name on the command line, not (chip, offset), but internally it resolves the names to (chip, offset) and then creates a request for each chip.  That is using the libgpiod v2 API, but the same mechanic would work with libgpiod v1.

Vernon-Naidoo-1, 2023-01-20:
Hi, sorry about posting this on the wrong project. You explanation was very helpful, thank you.

closed 2023-01-20 by Vernon-Naidoo-1