warthog618/go-gpiocdev #36: mac os undefined uapi.OutputValues

c-linse, 2023-02-22:
Hello

I want to create an application on my mac using this gpiod library and I am facing an compilation err.

thats my responsible code

```
if os.Getenv("BOARD") == "PI" {
		l, err := gpiod.RequestLine("gpiochip0", offset, gpiod.AsOutput(v))
		if err != nil {
			panic(err)
		}
		// revert line to input on the way out.
		defer func() {
			l.Reconfigure(gpiod.AsInput)
			l.Close()
		}()
}

```

And the resulting err:

`# github.com/warthog618/gpiod
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/options.go:93:50: undefined: uapi.OutputValues
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/options.go:101:34: undefined: uapi.LineConfigAttribute
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/gpiod.go:326:35: undefined: uapi.LineInfo
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/gpiod.go:351:37: undefined: uapi.LineInfoV2
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/gpiod.go:391:26: undefined: uapi.LineInfo
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/gpiod.go:401:28: undefined: uapi.LineInfoV2
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/gpiod.go:596:43: undefined: uapi.HandleFlag
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/gpiod.go:629:42: undefined: uapi.EventFlag
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/gpiod.go:642:49: undefined: uapi.LineFlagV2
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/gpiod.go:679:55: undefined: uapi.LineAttribute
../../../../go/pkg/mod/github.com/warthog618/gpiod@v0.8.1/options.go:93:50: too many errors`

Can someone explain me why this happens?

warthog618, 2023-02-22:
gpiod is Linux specific so you need to build your app for Linux, else go will try to build it for mac, and it will explicitly not build for mac as the underlying API does not exist on mac.

As mentioned in the Tests section of the README you can cross-compile for Pi using
```
GOOS=linux GOARCH=arm GOARM=6 go ...
```
That can be GOARM=7 for later Pis such as the Pi4.

c-linse, 2023-02-23:
Perfect thank you for the hint!

closed 2023-02-23 by c-linse