warthog618/go-gpiocdev #37: Dependency on GPL-3.0 library (go-udev)

ccollins476ad, 2023-07-25:
gpiod's mockup code uses the go-udev library (https://github.com/pilebones/go-udev), which is licensed as GPL-3.0. Applications that use gpiod don't actually include the mockup code, so technically there is no issue with the licensing here. However, some automated license checking software isn't sophisticated enough to differentiate between test code and the library itself. As a result, gpiod gets flagged as a GPL library.

I realize this is more of a "me" problem than an issue with gpiod, but I wanted to bring it to your attention. If there is some way to remove the go-udev dependency that would make it easier to use this library. I do think the code in `mockup` needs to have its license changed to something GPL compatible (header at the top of `mockup/udev.go` indicates MIT licensed).

warthog618, 2023-07-26:
I have a working branch (gpiocdev) that has already switched from using `gpio-mockup` for testing to `gpio-sim`, so that dependency is already gone there, though that branch is still a WIP and hasn't made it's way to a release yet.

It certainly be be part of the release when `gpiod` gets renamed `gpiocdev`.
One of these days I'll get around to it, but that day is not today.

warthog618, 2023-07-28:
I've cherry-picked the change from `gpio-mockup` to `gpio-sim` from the `gpiocdev` branch into master to remove the `go-udev` dependency, and tagged that as v0.8.2.
See if that works for you.

closed 2023-08-17 by warthog618