warthog618/go-gpiocdev #39: Create pin mappings for Milk-V Duo

pavelanni, 2023-09-03:
Hello Kent,

I received a Milk-V Duo board (RISC-V-based with Linux) recently and started using your library to work with GPIOs. I would like to add pin mappings for the board, but I need your advice on how to organize them properly. 

In this board there are 5 GPIO chips (`gpiochip0` to `gpiochip4`). The pins are named like `GPIOA14`, `GPIOC9`, or `PWR_GPIO21` and they are served by different chips. In this case the names above would be translated to `gpiochip0` + offset 14, `gpiochip2` + offset 9, and `gpiochip4` + offset 21 respectively.

I started thinking about the right way to translate pin names to proper `RequestLine` calls and it seems logical to me to create a type `gpio` that would contain both the chip name and the offset. In the `device` directory I would have a function that returns that type given a pin name. Then I'd call `RequestLine` with arguments like `(gpio.chip, gpio.offset)`

What would you say?

Thanks!

PS. Here is the board docs if you are curious: https://milkv.io/docs/duo/overview

warthog618, 2023-09-04:
Firstly, you don't have to follow anything in the device directory - you can perform your own mappings as you see fit.

If you do want to be reasonably consistent with the existing ```rpi``` and ```bananapi``` devices mappings, you should provide a Pin() that returns (chip,offset,error).  How you implement that function is up to you.  It could just be a literal mapping from name to (chip, offset), or it could do something algorithmic, depending on your line names and how you feel.

I don't see any point of creating a ```gpio``` type for the return value (chip,offset), and that should be called ```line_id``` or something anyway as it is not an actual GPIO - it is a line identifier.

You could also search for the line by name, as the ```gpiodctl find``` command does, assuming the lines are correctly named by your kernel/DT configuration (i.e. returned by ```gpiodctl info```).  But if your mappings are fixed then it is simpler to hard code them, as is done for the ```rpi``` and ```bananapi```.

pavelanni, 2023-09-04:
Thanks for your quick response, Kent!

> Firstly, you don't have to follow anything in the device directory - you can perform your own mappings as you see fit.

Yes, I figured out that I don't have to fork your whole repo and created one just for the mappings. Composition is better than inheritance, that's what I heard ;-) 

Here it is: https://github.com/pavelanni/gpiod-milkvduo

> I don't see any point of creating a gpio type for the return value (chip,offset), and that should be called line_id or something anyway as it is not an actual GPIO - it is a line identifier.

Good point about the naming -- I'll rename it.
I still think that I need a mapping between the string name (`GPIOA12`) of a pin and a pair (`gpiochip0`, 12). 

As for the compatibility -- I'll look into it. It seems for Milk-V Duo all the offsets are just numbers from pin names, so it's linear. 

> ... assuming the lines are correctly named by your kernel/DT configuration (i.e. returned by gpiodctl info).
Just checked -- almost all the lines in the `gpiodctl info` output are `unnamed`.
Looks like the kernel doesn't care much about it ;-) 

Thanks again!

warthog618, 2023-09-04:
I agree inheritance is overrated - composition is generally better.

> I still think that I need a mapping between the string name (`GPIOA12`) of a pin and a pair (`gpiochip0`, 12).

Sure, that is exactly what your mapping function will do.

> Just checked -- almost all the lines in the `gpiodctl info` output are `unnamed`.
> Looks like the kernel doesn't care much about it ;-)

The Pi used to be the same, which is one reason we had to rely on the (chip,offset) mapping.
The latest Pi kernels/DT configs now set all the line names.

pavelanni, 2023-09-04:
Thanks for you advice, Kent! I created a repo that includes the mapping function, a simple example, and the board info: https://github.com/pavelanni/gpiod-milkvduo. I hope somebody will find it useful.

I think this closes the issue.

closed 2023-09-04 by pavelanni