warthog618/go-gpiocdev #40: Debounce timeout does not change until reboot

benni-tec, 2023-09-22:
Hi Kent,
first of thank you for this amazing package!

When debuging my project I changed the timeout period on the `gpiod.WithDebounce(timeout)` Option. After a restart of my program (through Goland), the timeout did not change until reboot.

I don't know if this is related to how Goland kills programs (maybe missing cleanup?) but nevertheless this also might indicate that the debounce interval can not be changed while the program is running.

This could also be an OS issue, maybe the syscall is not correctly overriding the old interval?

**Device:** Orange Pi 5
**OS:** Ubuntu ([adapted](https://github.com/Joshua-Riek/ubuntu-rockchip) for rockchip cpus) 5.10.160-rockchip aarch64

I thought I would bring this to attention regardless, maybe you can reproduce the issue!

Regards, Benjamin

warthog618, 2023-09-22:
If your program is just requesting a line with debounce, and re-running it with a different debounce does not change the apparent debounce value then the problem is either in the kernel GPIO driver for your device, which I cannot test as I don't have an Orange Pi handy, or in how you are running your program from Goland (I don't use that either).

You could rule out the latter by calling ```gpiocdtl mon``` with different debounce periods (gpiodctl is in the cmd directory).

You can also call ```gpiodctl info``` while a program is running to check what the debounce period is set to:
```
$ ./gpiodctl mon -d 20ms gpiochip0 18 &
$ ./gpiodctl info
...
	line  18:      GPIO18     gpiodctl-mon [input,used,debounce-period=20ms]
...
```

warthog618, 2023-10-01:
Without any further input I can only assume that the underlying problem is either the way the program is called or the device driver, not gpiod.  So closing the issue as there nothing more I can do here.

closed 2023-10-01 by warthog618

benni-tec, 2023-10-13:
Sorry for the late reply. 
Yes I agree, the problem probably lies with the driver (or ubuntu-rockchip) since it realy is a niche OS/usecase! 

Since I am time constrained at the moment and a change of the debounce time is not needed for my usecase, I did not investigate this any further. It seems like this is not an issue with other setups, so it probably is a driver issue.

Nevertheless thank you!