warthog618/go-gpiocdev #43: Reading back value always returns 0

pascalhuerst, 2023-10-16:
I can setup a gpio and control the output with `SetValue(v)` without any issues, but reading back the value always returns 0:

```
...
line, err := c.RequestLine(o, opts...)
...
line.SetValue(0) // works, gpio goes lo
line.SetValue(1) // works, gpio goes hi
v, err := line.Value() // err is nil, but v is still 0
```

I'm quite certain this worked before. We are currently using v0.8.2.

warthog618, 2023-10-16:
If you are quite certain it worked before then what changed?

pascalhuerst, 2023-10-17:
Fair enough. I admit I was being lazy. We have not changed anything in our user space code in the last 2 years or so, other than updated this library. We are however always very much up-to-date with the kernel, so it is quite possible something broke there. I will debug the driver next time I have hardware at hand or see what results I get with the c implementation.

closed 2023-10-17 by pascalhuerst

pascalhuerst, 2023-10-17:
> A reasonably recent `strace` will also decode the GPIO ioctls, so you could use that to confirm that `gpiod` is propagating the value returned by the kernel.
> 
> e.g.:
> 
> ```
> $ strace --trace=ioctl gpiodctl get -a gpiochip0 14
> --- SIGURG {si_signo=SIGURG, si_code=SI_TKILL, si_pid=1192, si_uid=1000} ---
> ioctl(3, GPIO_GET_CHIPINFO_IOCTL, {name="gpiochip0", label="pinctrl-bcm2711", lines=58}) = 0
> ioctl(3, GPIO_V2_GET_LINEINFO_IOCTL, {offset=0} => {name="ID_SDA", consumer="", flags=GPIO_V2_LINE_FLAG_INPUT, num_attrs=0}) = 0
> ioctl(3, GPIO_V2_GET_LINE_IOCTL, {num_lines=1, offsets=[14], consumer="gpiodctl-get", config={flags=0, num_attrs=0}} => {fd=7}) = 0
> ioctl(7, GPIO_V2_LINE_GET_VALUES_IOCTL, {mask=0x1} => {bits=0x1}) = 0
> 1
> +++ exited with 0 +++
> ```
> 
> The value is in the bits bitmap of the GPIO_V2_LINE_GET_VALUES_IOCTL, and in this case is 1.

Argh, good point, should have done that from the beginning :facepalm: 

```
ioctl(3, GPIO_V2_GET_LINEINFO_IOCTL, {offset=4} => {name="sublink-1-ipal-mute", consumer="", flags=GPIO_V2_LINE_FLAG_OUTPUT, num_attrs=0}) = 0
ioctl(3, GPIO_V2_GET_LINE_IOCTL, {num_lines=1, offsets=[4], config={flags=GPIO_V2_LINE_FLAG_OUTPUT, num_attrs=1, attrs=[{values=0, mask=0x1}]}, consumer="gpiod-2049"} => {fd=7}) = 0
ioctl(7, GPIO_V2_LINE_SET_VALUES_IOCTL, {bits=0x1, mask=0x1}) = 0
ioctl(7, GPIO_V2_LINE_GET_VALUES_IOCTL, {mask=0x1} => {bits=0}) = 0
```

This sets the value to `1` and reads it back. Kernel returns `0`, so the issue is in the driver. Sorry for the noise!