warthog618/go-gpiocdev #44: Determine chip from watcher callback

BGodding, 2023-10-30:
I am working on an application that can be configured on runtime to send GPIO events to an external service. 

I would like to use a single generic callback for all line change events across multiple chips, but as far as I can tell `LineEvent` and `LineInfoChangeEvent` do not provide a way in which to determine the source chip of the event, only the index of the line. I think the only workaround I found was to set a unique consumer for each chip and read that from the `LineInfo` and line names are unset.

Would there be any hesitation to adding chip name to the event change callback structs? I would be happy to submit a PR if there is not opposition.

warthog618, 2023-10-30:
Most callbacks do not require the chip so adding it in all cases is wasteful.
I wouldn't use a single generic callback, I would provide a callback for each request/chip that decorator the event with whatever additional info you want and passes it on to your external service.

BGodding, 2023-10-30:
Thank you for the reply, potential performance impact was the drawback I saw as well.

The issue is I don't know prior to compilation time how many chips are present. Could be 1 or 3 or more. I could create a large number of callbacks and hope that covers any amount of chips I run into but that did not seem idiomatic.

warthog618, 2023-10-30:
Another issue is that you want to decorate with chip name, but someone else might want line name or their own identifier.

I would dynamically create a closure that performs the decoration and then calls your single generic callback.  That would be per chip (for ```LineInfoChangeEvents```) or per line request (for ```LineEvents```).  That does not require foreknowledge of the number of chips or line requests and seems more idiomatic to me than insisting on a single generic callback.

BGodding, 2023-11-01:
That is a great idea and what I ended up doing. Cheers.

closed 2023-11-01 by BGodding