warthog618/go-gpiocdev #46: Releases 0.9 and 1.0

Klaus-Tockloth, 2024-01-12:
Just a question not an issue: Is there a rough schedule for the releases 0.9 and 1.0?

PS: Great project !

warthog618, 2024-01-12:
No schedule for 0.9, just when I get the time - and other things seem to keep cropping up.
And no plan for 1.0 until 0.9 is out.

warthog618, 2024-03-24:
0.9 is out, the primary changes being the rename to gpiocdev and the splitting out of the cli into a separate repo.
There are a few minor additions that I might put into a 0.9.1.
I don't foresee any major changes before 1.0, but will see how 0.9.x settles first.

warthog618, 2026-10-04:
As per #60, I'm now considering a 0.10 series that will drop uAPI v1 support, cos as this point it is just pointless cruft.
That would then form the basis of 1.0.