warthog618/go-gpiocdev #47: Permission Denied on only SetValue()

Gnyblast, 2024-05-30:
I have the following code for opening and closing GPIO pins. I double checked the pins I'm using, they are for sure Output pins and I'm configuring them as output at the start of the program by also setting them defautl to 0 and `asActiveHigh`. I checked my permissions on the Raspberry and I'm the god in the machine. I can easily use `pinctrl set` command with sudo or without sudo to change the pin states and confirm it with `raspi-gpio get` that they are changed.

My problem is, when my app tries to set pin value to 1 or 0, I get `permission denied` from `line.SetValue()` method. I tried running my program as sudo in the raspberry or as root but made no difference, that's why I'm thinking there might be a bug in the code or I might be missing something. Here is my function:

```go
func (g *GPIOEngine) toggleIOPin(chipname string, pinNumber string, toggle int) error {

	pin, err := rpi.Pin(pinNumber)
	if err != nil {
		g.logger.Debug("Failed to get pin")
		return err
	}

	line, err := g.chips[chipname].RequestLine(pin)
	if err != nil {
		g.logger.Debug("Failed to request line")
		return err
	}

	defer line.Close()

	err = line.SetValue(toggle)
	if err != nil {
		g.logger.Debug("Failed to set value")
		return err
	}

	return nil
}
```

and this is how I initialize `g.chips` to make things clear for you:
```go
	for _, chipName := range gpiocdev.Chips() {
		logger.Debugf("Chip: %s Initialized", chipName)
		chip, err := gpiocdev.NewChip(chipName)
		if err != nil {
			return nil, err
		}

		chips[chipName] = chip
	}
```

warthog618, 2024-05-30:
Setting the value on a pin not explicitly requested as an output results in the `permission denied` error you are seeing.
You need to request the line as an output to be able to set the value, so oass the `gpiocdev.AsOutput(v)` option to `RequestLine()`.

And checkout the examples - [toggle_line_value](https://github.com/warthog618/go-gpiocdev/blob/master/examples/toggle_line_value/main.go) would be the most relevant, particularly this bit:

```go
	v := 0
	l, err := gpiocdev.RequestLine(chip, offset, gpiocdev.AsOutput(v))
```

Gnyblast, 2024-05-31:
Hey there again,

Thanks for the answer. But I wanted to insisnt on my implementation because it was making much sense for my code, so I had to dive in your code base a bit to find out this:

```go
	var option gpiocdev.OutputOption = []int{2}
	line, err := g.chips[chipname].RequestLine(pin, option)
	if err != nil {
		g.logger.Debug("Failed to request line")
		return err
	}
```
So if you pass `var option gpiocdev.OutputOption = []int{2}` as an option to a `RequestLine` method of a `Chip` struct it works. Even though it still works, It looks a bit redundant to me to re-set the output option when requesting a line, since the line was reconfigured at the start as output and in `l.info[0].Config.Direction` knows that it's output. But for some reason it uses `l.defCfg.Direction` in the `SetValue` method of `Line` struct and looks like `l.defCfg.Direction` is not beign set properly on `chip.RequestLine()` method.

On the otherhand, `info` element of `baseline` is not also being initialized on start, so makes `l.info()` unusable  in the `SetValue` output if check.

I still left it not initialized at start, which would be better maybe but instead I injected the usage for `l.info()` into `SetValue` method of `Line` and `Lines` structs that solves this problem.

warthog618, 2024-05-31:
If you insist then I need to explain further...

It is required by the kernel that the line is requested as an output before you can set it.  This is to demonstrate that you know that is what you want, and not allow you to go messing with output lines by accident.
If you request the line as-is then you can still read the current value before reconfiguring the line as an output to change that value.

Sorry if that feels redundant to you, but that is the way it is and I'm not about to change it.

Gnyblast, 2024-05-31:
I see your point and it's a valid case. I'm also not planning to fork outside of the project since I can use `var option gpiocdev.OutputOption = []int{2}` on `RequestLine()` but I just wanted to mention because in my own mentality, I thought like, if you can reconfigure, why you should re-declare or if you should re-declare why you reconfigure.

Thanks for the answer. Have a great day.

Gnyblast, 2024-05-31:
I also wanted to mention this,

Since you already have the if check in `SetValue` method, if the line is configured as output, and even if you accidentaly mess with the lines, worst case scenario you will get a "Permission denied" error as I did.

I suggest you the have a look to my merge request, because the output check is still there and you still can explicitly declare it on `RequestLine` but it also respect the current configuration of the line itself.

warthog618, 2024-05-31:
I saw it, and I'm not about to do things automagically when it comes to playing with hardware.  It you want it to be an output then you need to explicitly say so.  Then if the magic smoke comes out it is your fault ,not mine.

Your use case is handled, just not as simply as you would like.  If that bothers you then wrap it to behave the way you want.

Gnyblast, 2024-05-31:
That's fair, thanks.

warthog618, 2024-05-31:
Btw, your code feels redundant to me - why have to set the value separately from requesting the line?  That requires two ioctl calls, whereas using the AsOutput option requests it and sets it in the one ioctl.

Gnyblast, 2024-05-31:
> Btw, your code feels redundant to me - why have to set the value separately from requesting the line? That requires two ioctl calls, whereas using the AsOutput option requests it and sets it in the one ioctl.

Well I though initializing the chips at start and setting them in a point map would make it faster since don't need to create chips everytime, then request line but instead just request the line.

```go
	logger.Debug("Initializing GPIO Chips")
	var chips map[string]*gpiocdev.Chip = make(map[string]*gpiocdev.Chip)
	for _, chipName := range gpiocdev.Chips() {
		logger.Debugf("Chip: %s Initialized", chipName)
		chip, err := gpiocdev.NewChip(chipName)
		if err != nil {
			return nil, err
		}

		chips[chipName] = chip
	}

	err := resetOutputPinLinesStates(chips, outputDevices, logger)
	if err != nil {
		return nil, err
	}

	err = resetInputDevicesLinesStats(chips, inputDevices, logger)
	if err != nil {
		return nil, err
	}
```

warthog618, 2024-05-31:
> Well I though initializing the chips at start and setting them in a pointer map would make it faster since don't need to create chips everytime, then request line but instead just request the line.
> 

Whether you re-use the chip is a separate issue.  Requesting the line from the chip and setting the value are two ioctl calls, whereas requesting it as an output is only one.

If you are making multiple requests on the same chip then it is slightly quicker to open the chip once and use it to create all the requests.  But you don't need the chip once you have the requests, and if you are only creating the requests once at the start of the program then it might not be worth the effort, which is why the library also provides the top-level RequestLine().

Gnyblast, 2024-05-31:
Yeah ok, I already fixed my code to this which is single request from the `chip.RequestLine()` to get line and set value.
```go
	var outputOption gpiocdev.OutputOption = []int{2}
	var activeStateOption gpiocdev.OutputOption = gpiocdev.AsOutput(activeState)
	line, err := g.chips[chipname].RequestLine(pin, outputOption, activeStateOption)
```

But yes I need the chip to be there, because application is a server and occasionally accessing the chips which would make it faster this way.

warthog618, 2024-05-31:
What do you need the chips for, other than requesting the lines?  Keeping the chips open will be slightly faster, but unless you are making heavy use of the chips then you probably wouldn't notice.

> ```go
> 	var outputOption gpiocdev.OutputOption = []int{2}
> 	var activeStateOption gpiocdev.OutputOption = gpiocdev.AsOutput(activeState)
> 	line, err := g.chips[chipname].RequestLine(pin, outputOption, activeStateOption)
> ```

What is this code trying to do?  What is outputOption for? What is wrong with just:

```go
	line, err := g.chips[chipname].RequestLine(pin, gpiocdev.AsOutput(activeState))
```

Gnyblast, 2024-05-31:
Sorry forgot that `gpiocdev.AsOutput()` had implicit pin set to output. Doing that.

warthog618, 2024-05-31:
Not sure if you've run across [functional options](https://uptrace.dev/blog/golang-functional-options.html) before, but the options to RequestLine() are just that - options.  They don't do anything themselves, they tell RequestLine() how to behave.

There is no reason to construct an option yourself - that is what the helper functions do, so AsOutput() creates an OutputOption which is passed in to RequestLine().  So AsOutput() isn't doing anything implicit - it is telling RequestLine() explicitly that you want to request that line as an output, and it provides the value to set it to.

Btw, while the kernel does allow you to request a line as output without specifying a value, the kernel interprets that as setting it to 0.  And, as a value is always set, the library requires a value when requesting the line as output - so it isn't implicitly set it to 0 - if you want to set it you need to specify a value.

Gnyblast, 2024-05-31:
I understand, I meant implicit, relative to my code, it might be explicit within the library ofc.

My final code looks like this:
```go
func (g *GPIOEngine) SignalPinToOpen(chipname string, pinNumber string) error {
	g.logger.Infof("Opening pin %s on chip %s", pinNumber, chipname)
	return g.toggleIOPin(chipname, pinNumber, OPEN)
}
func (g *GPIOEngine) SignalPinToClose(chipname string, pinNumber string) error {
	g.logger.Infof("Closing pin %s on chip %s", pinNumber, chipname)
	return g.toggleIOPin(chipname, pinNumber, CLOSE)
}

func (g *GPIOEngine) toggleIOPin(chipname string, pinNumber string, activeState int) error {

	pin, err := rpi.Pin(pinNumber)
	if err != nil {
		g.logger.Debug("Failed to get pin")
		return err
	}

	line, err := gpiocdev.RequestLine(chipname, pin, gpiocdev.AsOutput(activeState))
	if err != nil {
		g.logger.Debug("Failed to request line")
		return err
	}

	defer line.Close()
	return nil
}
```
I also eliminated chip initialization and delived all to `RequestLine()` to handle.
Thanks for all the helps, I think I'm good now.

closed 2024-06-01 by warthog618