warthog618/go-gpiocdev #51: Go Version

ea7kir, 2024-07-18:
Please can you update to a more recent version of Go?
You're using 1.19, the current version as I write is 1.22.5.

warthog618, 2024-07-18:
What do you mean by "using"?
The go.mod does not lock you into a specific version - that is the minimal version.
Builds fine for me with 1.22.5, so not sure what your issue is.

ea7kir, 2024-07-18:
Hi, and phakyu for responding.

Yes, it does compile and work with Go latest.   The issue is with at least one of the requirements.

Your release brings in golang.org/x/sys v0.18.0, but the latest is v0.22.0. 

Also, you are not using current log an error packages.

I use your package and am really grateful, but I do worry how long it will continue working, especially as I intend to upgrade from Bullseye to Bookworm.

Regards,
Michael Naylor - EA7KIR
https://michaelnaylor.es

warthog618, 2024-07-18:
And that is a separate issue unrelated to the Go version.

You do not seem to understand how Go dependencies work.
The require statement in the go.mod specifies the minimum required version of a module.
Your build is not locked to that version.
If you need a later version then specify it in your other dependencies and the [version selection algorithm](https://go.dev/ref/mod#minimal-version-selection) should select it, as it would not conflict.
That existing version will not magically stop working.  If there is a fatal flaw in that particular module version then it can be unpublished by the author and the build will move to another later version.

And what does any of that have to do with your distro version?
I have upgraded to Bookworm and that changed nothing relevant to the Go build.

My worry is that you worry for nothing.
So again, what is your actual problem?

warthog618, 2024-07-18:
Oh, and if you are coming from a Pi environment and are concerned that the upgrade from Bullseye to Bookworm will break things, as it did for so many Pi specific libraries or libraries using the GPIO sysfs interface, then fear not as this library uses the GPIO character device interface that still works fine unchanged, and will continue to do so for the foreseeable future.
The only proviso being that if you are upgrading to a Pi5 you will need to switch from using _gpiochip0_ to _gpiochip4_.

closed 2024-07-22 by warthog618