warthog618/go-gpiocdev #52: undefined some uapi package

Jaxx1911, 2024-07-20:
[image]
Why i have this error when trying to use your lib :(( Pls help

warthog618, 2024-07-20:
I can't even tell what you are doing, so no idea.
What did you do to produce that?
And don't post images - cut and paste text instead.

warthog618, 2024-07-20:
Ahh, are you building on Windows or Mac?  This library is Linux specific. Check the [installation instructions](https://github.com/warthog618/go-gpiocdev#installation) if you are having problems during installation. e.g.

```
go get -d github.com/warthog618/go-gpiocdev
```

And if building on a non-Linux platform you need to setup the cross-compilation variables, e.g.

```
$ GOOS=linux GOARCH=arm GOARM=6 go build
```

Jaxx1911, 2024-07-20:
Thanks u so much :33

closed 2024-07-20 by Jaxx1911