warthog618/go-gpiocdev #54: Handler Library for Input Sensors

Gnyblast, 2024-08-01:
Do you know of any library that I can be used with your `gpiocdev.WithEventHandler(handler)` as a handler that provides methods for each sensor? For example I might watch an input line with a DHT11 sensor on it but all I'm going to get is meaningless 0s and 1s. In order to make some sense out of these, I guess I need a handler that can make some sense out of these 0s and 1s to build up a binary message that can be converted then.

I found many libraries but almost all of them has to do their own `for` loop to watch the pin and the way they watch is some different library. This is some how sounds inconsistent to me, maybe I'm misunderstanding or skipping something with them or with your library, I don't know, just wanted to ask...

warthog618, 2024-08-02:
No I don't know of any, and I'm not familiar with the DHT11.

From the datasheet, you should be able to decode each bit of data from the DHT11 by checking how long the line is high.  For the handler that would mean the length of time between a rising and falling edge.  The bit duration can be calculated using the timestamps in the events.

warthog618, 2024-08-02:
There is a [kernel IIO driver](https://github.com/torvalds/linux/blob/master/drivers/iio/humidity/dht11.c) for the DHT11.  Have you considered using that?

Gnyblast, 2024-08-02:
Yes actually kernel has a lot of drivers for different sensors, DHT11 was just one of them. I saw that implementations in many places where kernel writes into a file in  `/sys/bus/` directory general and with go they just read and parse the file to get the data.

This is probably one of the best way of reading data from this type of sensors because it seems like they are open to the error a lot of time and kernel drivers might be the most stable option here but... It doesn't suit me well because I'm building a general purpose software and I expect end users to be able to do stuff without needing extra configurations on their RPI, that's why I wanted to go to code level reads.

Yeah as you said there are ways for reading these HIGHs and LOWs from the input and make it meaningful, I was just wondering if there was any library already doing this as a handler for your program, I think there isn't.

I might better spend some time and create my own handler library for the sensors I'm gonna need.
Thanks.

closed 2024-08-02 by Gnyblast

Gnyblast, 2024-08-02:
I started writing my own handlers library, nothing tested yet, going theoretically since I don't seem to have most of the sensors but I'll write test and test physically as I get sensors.

https://github.com/Gnyblast/go-gpiocdev-eventhandlers