warthog618/go-gpiocdev #56: Proposal: *gpiocdev.Lines should allow the developer to flip specific bits without knowing prior state

fasmide, 2024-09-06:
Hi

It would be very nice to be able to set specific bits on a `*gpiocdev.Lines` structure. 

Currently, `*gpiocdev.Lines` only provides a `SetValues(values []int) error,` which in turn requires us to know the prior state even if we want to change only one specific. 

At the moment, I'm doing something like 
```
var bitmap []int
_ := lines.Values(bitmap)

// do some changes
bitmap[5] = 1
_ := lines.SetValues(bitmap)
```

While working, I think the above code is racy if some other goroutine changed stuff between `Values()` and `SetValues()`. 

I'm not sure what such an API should look like, but it could leverage the already exported `uapi.LineValues`, which contains both a mask and a bitmap. The developer could then use something like:
```
	bits := uapi.NewLineBitMask(8)
	bits.Set(4, 1) // fourth bit should be active

	mask := uapi.NewLineBitMask(8)
	mask.Set(4, 1) // only the fourth bit should be changed

	_  := lines.SetLineValues(uapi.LineValues{bits, mask})
```

Please let me know if there's a better / already implemented way of manipulating single bits in a gpiocdev.Lines collection

warthog618, 2024-09-06:
What is the use case for sharing a multi-line request between goroutines?
Might separate requests be more appropriate?

warthog618, 2024-09-06:
Oh, and what you are doing by reading the current Values() is not the way to do it.  You should remember the values set when requested, and modify those and apply using SetValues().  Sharing a request between goroutines is an anti-pattern.  Have one goroutine manage the request and have other goroutines message THAT goroutine to access the request. i.e. Don't communicate by sharing, share by communicating.

fasmide, 2024-09-06:
The thing about the example code being racy was just a side note 

> You should remember the values set when requested, and modify those and apply using SetValues().

hmm okay if that's the case I guess ill have to manage the state elsewhere - its just that I have no need for it other then having to change single bits here and there - so I thought maybe this functionality could be provided by the `*gpiocdev.Lines`.

warthog618, 2024-09-06:
That would have to store state and provide locking or a managing goroutine.  This is additional overhead for those not wanting to share the request between goroutines.  And it is not idiomatic Go.

What is your use case that requires setting lines on the same request from separate goroutines?

fasmide, 2024-09-06:
> That would have to store state and provide locking or a managing goroutine. This is additional overhead for those not wanting to share the request between goroutines. And it is not idiomatic Go.

That's why I thought it would be nice to be able to provide a mask for `SetValues()` - I would imagine no state or locking should be required in that case.

anyways, what I'm currently building is a web server that toggles bits on a gpiochip - and yes, you are right, these requests do execute in different goroutines

warthog618, 2024-09-06:
> That's why I thought it would be nice to be able to provide a mask for `SetValues()` - I would imagine no state or locking should be required in that case.
> 

Using uapi v2 that would be possible, but for uapi v1 you always need to set all lines, so that would need to be stored and protected with a lock.

> anyways, what I'm currently building is a web server that toggles bits on a gpiochip - and yes, you are right, these requests do execute in different goroutines

That does not explain why you need to have separate goroutines messing with lines on the one request.

fasmide, 2024-09-06:
> anyways, what I'm currently building is a web server that toggles bits on a gpiochip - and yes, you are right, these requests do execute in different goroutines

Sorry, by toggle, i mean write explicit values - toggling would, of cause require me to know the prior state :)

warthog618, 2024-09-06:
> Sorry, by toggle, i mean write explicit values - toggling would, of cause require me to know the prior state :)

Surely the user want to know the current value so you need to store the state anyway.  And you should not rely on reading the value of an output - you should store it yourself - as some hardware/device drivers don't return the last set value.

fasmide, 2024-09-06:
> Using uapi v2 that would be possible, but for uapi v1 you always need to set all lines, so that would need to be stored and protected with a lock.

Yikes, i didn't know that - I see why it might not be a good idea in that case

> That does not explain why you need to have separate goroutines messing with lines on the one request.

The current flow of my process is something like

main does a RequestLines on the whole gpiochip and stores it, defines some http handlers and starts the webserver.
then http requests comes in, the handlers uses the `*goipcdev.Lines` to write bits - and to my knowledge, these are executed by the standard library web server in separate goroutines

warthog618, 2024-09-07:
The API is the way it is to avoid the need for additional state and locks, while still being compatible with both versions of the uAPI where possible.  You can add a wrapper with more convenient functions yourself if that is what you want, but in your case you should reconsider your design.

You have no need to request all the lines in the one request.  In fact that doesn't always work.
What if the chip has more than 64 lines? Requests are limited to 64 lines, but chips are not.

Requesting ALL the lines would force all userspace access to those lines through your server, as only one process can request a line at a time.  That doesn't play nice with other apps using GPIO.
What if lines have already been requested?

warthog618, 2024-09-07:
Oh, and if you are requesting all the lines as outputs that could be bad.

fasmide, 2024-09-07:
All I'm asking is a maskable way of setting bits, just like the uapi V2 provides to everyone else 

I can see how such a thing is somewhat not possible to give users with older kernels, but I don't know - maybe we could return an ErrUnsupported or something

warthog618, 2024-09-07:
I know.  And I question if that is the right way to go in Go.  But do keep asking.

If I were to add a function to set a subset of lines it would take a dict mapping offsets to values, so you could set multiple lines, not just one.

warthog618, 2024-09-07:
To be clear, the multi-line request is intended for bit-bashing busses so you can set a number of lines simultaneously.  It is not there just to bundle a bunch of lines for the hell of it.  For your use case separate requests make more sense.

fasmide, 2024-09-09:
Well, I appreciate the discussion on the topic, and while I eventually choose another route (requesting lines individually, as suggested), I still believe it is a shame the uapi v2 provides a maskable way of writing bits to a collection of lines - and that we can't have this convenience.

Feel free to reopen this issue, should you feel the same :)

closed 2024-09-09 by fasmide

warthog618, 2024-09-09:
I don't have a solution I'm happy with.  A function that only sets a single line of a multiline request is pretty niche, and probably indicates you shouldn't be using multiline in the first place.  A more general solution is to accept a subset of lines to set, but I haven't come up with an API that is any better at that than the uapi v2 call, uapi.SetLineValuesV2(), itself.  There is nothing stopping you using that directly, so I'm not sure what a v2 specific function in gpiocdev that mirrors it would add.

Were I rewriting the API just for v2 that might be different, e.g. SetValues() currently takes the lines in the order requested, so the caller needs to remember the order.  I might be inclined to change that to a map from offset to value, though then the caller needs to be making maps or lists of (offset,value) pairs, all of which is a bit awkward, so swings and roundabouts.